witnessmenow/ESP32-Cheap-Yellow-Display #27: Feature Request: ESPHome + LVGL Examples

RyanEwen, 2023-08-19:
I love what's being done here and have a request. I am trying to learn Both ESPHome and LVGL for use on one of these devices. Could someone come up with a simple example that uses these together?

So far I've found this: https://github.com/fvanroie/esphome-lvgl which has helped me compile something that the board accepts but I've yet to get anything showing on the screen

ncmreynolds, 2023-11-15:
I have added info about making the LVGL examples work in #55 hopefully Brian will merge it.

witnessmenow, 2023-11-18:
An example that has both LVGL and ESPHome seems like it might be too specific

Misiu, 2024-09-18:
That would be awesome, especially that ESPHome has LVGL native support: https://esphome.io/components/lvgl/

sickrandir, 2024-10-22:
I keep finding hints on reddit that people got LVGL running on CYD with ESPHome but I can't find a complete yaml example. It would be really awesome to have one.

RyanEwen, 2024-10-22:
I forgot about this request..

I've been using lvgl on e different Sunton devices and an Elecrow now. 

For the 2.8 specifically:
https://community.home-assistant.io/t/esp32-2432s028r-sunton-2-8-cyd-config/780600/21

I don't actually have any lvgl widgets in the example configs I post to the ESPHome forum but I am slowly working on some in a GitHub repo here: https://github.com/RyanEwen/esphome-devices

sickrandir, 2024-10-23:
> I don't actually have any lvgl widgets in the example configs I post to the ESPHome forum but I am slowly working on some in a GitHub repo here: https://github.com/RyanEwen/esphome-devices

Loving this. Thanks a lot, I'll follow your repo!

RyanEwen, 2024-10-26:
Rather than make a simplified demo layout I ended up just uploading the actual project that is in-use at my house. This is because how I organize code and files and modularize things has been evolving rapidly, resulting in many time-consuming sweeping changes. Rather do everything twice I figured I'd just share the in-use project and one day when I've settled on the best way to do everything I can make a demo or sample layout that is simpler :)

RyanEwen, 2024-10-26:
I should also mention that the ESPHome docs now have examples in them which might also be useful. 

https://esphome.io/cookbook/lvgl.html