witnessmenow/ESP32-Cheap-Yellow-Display #40: ESPHome screen on templates might be wrong

yeyeto2788, 2023-11-03:
Hey! Thanks for sharing this. 

I have tested the ESPHome templates and there seems to be an error on the display used. I changed the model to be `ILI9342` as the `ILI9341` did not work on my board (flickering a lot and not complete fill of the screen).

I also noticed that even when changing the model the color order instead of **RGB** the `ILI9342` seems to be **BGR**.

Let me know if this is something you'll want me to add in the templates.
Cheers.

witnessmenow, 2023-11-03:
Do you have the cyd with two USB ports by any chance? 

@jonnybergdahl is the person who contributed this example, so maybe he has some input

yeyeto2788, 2023-11-03:
Yes, I do. One is USB type C and the other is USB micro B (I think). Want me to share a picture of the board? Is there a way to check the hardware connected to the board?

BTW, I have a template combining most of the examples into one single template, is that something you'll want me to create a PR with?

witnessmenow, 2023-11-03:
Yes it seems to be a slightly different revision, I'm unclear what the connection is to the original cyd if it's a knockoff or a new version, but it's display is definitely a little different. So I'd imagine the existing config works for the single micro USB CYD, but your config is needed for this two USB version.

You'll have to excuse my ignorance, but I really know nothing about esphome yet so it's I don't have any opinion on combing the template

What are the advantages of it?

witnessmenow, 2023-11-03:
Raised #41 for a wider discussion around this type of CYD

yeyeto2788, 2023-11-03:
The advantage of having a complete template is that you can test various features (LDR, web server, RGB, display, etc) by compiling a single template. 

I'm not an expert either BTW I just tested it 😛

yeyeto2788, 2023-11-03:
I have created a PR with the template just in case you may want to add it into the repo https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/pull/42

yeyeto2788, 2023-11-03:
Taking a closer look at the board I noticed another difference, my board does not come with an external flash memory chip (unpopulated on the board)

Grey-Lancaster, 2024-01-17:
Just got one (CYD) esphome ili9342 does work better 9341 did not fill the screen and flickerd,  Only done hello world so far but ready to expand !

Grey-Lancaster, 2024-01-17:
Any idea why image is just using 2 colors?
```
image:
  - file: http://192.168.1.50:8123/local/dad2.PNG
    id: my_image
    resize: 120x120
    type: RGBA
```
```
display:
  - platform: ili9xxx
    model: ili9342
    spi_id: tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    rotation: 90
    color_order: RGB
```

yeyeto2788, 2024-01-22:
@Grey-Lancaster try setting the `color_order` to GRB as the ILI9342 seems to have the color order inverted. I have never used that property but if it is helpful this is what I'm currently using:

```yaml
display:
  - platform: ili9xxx
    id: esp_display
    model: ILI9342
    spi_id: spi_tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    rotation: 270
```

Grey-Lancaster, 2024-01-22:
Thanks, this works for me "basically" for text, but images are bland.  Actually I think the contrast may be too high ?

```
display:
  - platform: ili9xxx
    model: ili9342
    spi_id: tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    rotation: 270
    color_order: RGB
    id: "my_display" 
```