witnessmenow/ESP32-Cheap-Yellow-Display #41: Name the two USB CYD and handle it in documentation and examples

witnessmenow, 2023-11-03:
The CYD with two USB ports is a bit of an issue at the moment. It has a slightly different screen and the config and examples don't work right on it.

I think it's a knockoff rather than a newest version as the AliExpress store of the manufacturer still seems to sell the version with micro USB.

I believe it's the one great Scott linked to, so there are probably plenty of them floating around

It has the same name so despite being different, I think the best way to proceed is to document the difference and adapt instructions and examples to handle it

Any name suggestions?

"Cyd double USB" is pretty descriptive so might be good

witnessmenow, 2023-11-03:
I would also love to get some help with this as I don't have one

yeyeto2788, 2023-11-03:
What about CYD2U? hahaha, I know you wanted to keep a short name but seems like the more board variants the more letters will need to add.

Pinout seems to be exactly the same as I have tested some templates from the repo and everything works except for the display which is a different model `ILI9342`

If you need something related to this board I can help as I own one.

dieseltravis, 2023-11-04:
How about just `CYD2`?

witnessmenow, 2023-11-05:
I think the issue with anything in with 2 in the name sounds like a successor or a new revision, my gut feeling at the moment is it's a knock off

yeyeto2788, 2023-11-07:
It would be great to define what is needed no matter what it will be called afterward. If anything is needed on my end ping me. 

P.D: I would expect people not to care about the name and more about the code working and since not everything will work having the updated documentation is critical to avoid having several issues opened.

woodat, 2023-11-08:
Hello, working on a micropython implementation of the CYD and was using your repo as reference.   Found a few issues and thought I'd mention it in here.

If you're doing a bit of work on documenting different versions, there's a capacitive touch CYD with a neopixel and USB-C as well.  It's classified on Aliexpress as ESP32-2432S024 rather than ESP32-2432S028R.  Critically though (and something that took me all last night to troubleshoot) the Backlight on that version is on IO27, not IO21 (listed in the documentation here).  The RGB pins are also 4, 16, and 17 if that helps.

Also didn't know if I missed it, but if not it may be worthwhile to list out the hardware components somewhere in the documentation along with the models.  For example, I only knew it was an ILI9341 by going into your code :)

gjashley, 2023-11-08:
I have one labeled  ESP32-3248S035C (capacitive, 3.5"), and that has the backlight on GPIO27 too. It also has some other peculiarities which may lead me to shelve it for a while e.g. that ESPHome can't drive the full screen, and some forums saying you need to remove/bridge a resistor to get i2c to work.

yeyeto2788, 2023-11-08:
> Hello, working on a micropython implementation of the CYD and was using your repo as reference. Found a few issues and thought I'd mention it in here.
>
> If you're doing a bit of work on documenting [...]

@woodat, @witnessmenow  I think there is something related to the naming convention IMHO

| Board name | Pixels | Screen size | Touch type |
| --- | --- | --- | --- |
| ESP32-2432S024 | 2432 = 240x320 pixels |  024 = 2.4 inch LCD | N/A |
| ESP32-2432S028R |  2432 = 240x320 pixels |028 = 2.8 inch LCD | R = Resistive |
| ESP32-3248S035C | 3248 = 320x480 pixels | 035 = 3.5 inch LCD | C = Capacitive |

> I have one labeled ESP32-3248S035C (capacitive, 3.5"), and that has the backlight on GPIO27 too. It also has some other peculiarities which may lead me to shelve it for a while e.g. that ESPHome [...]

@gjashley I would recommend you try any example from ESPHome in the repo (https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/blob/main/Examples/ESPHome/1-Helloworld/yellowtft1.yaml) and try any of the models [here](https://esphome.io/components/display/ili9xxx#models)

I think the one with a size of 3.5 inches is the `ILI9488`

gjashley, 2023-11-08:
@witnessmenow > Agree that the primary focus should be ESP32-2432S028R (single and double versions) ... there are too many clones out there, and in some instances the yellow PCB seems to be the only material thing they have in common!?! I ordered one from your affilate link and hope to be making more meaningful contributions when it arrives

@yeyeto2788 > yes, I too see the pattern in the names.  
I tried all the displays listed on the [ili9xxx platform on ESPHome](https://esphome.io/components/display/ili9xxx.html?highlight=ili9xxx), and only the ili9341 and ili9342 worked, and even then at 240x320 (with snow on the rest of the screen, even using 'it.fill'). Using ili9488 or any of the other models, or trying to hardcode the resolution with the 'dimensions' setting threw up the error " DC Pin: GPIO2 => Failed to init Memory: YES!"

wanderling, 2023-12-13:
I vote `CYD+C` for the added USB-C port.

jaymcd, 2024-01-24:
I have a screen with the ESP32-3248S035C. I was trying to instal CYD-Klipper and bricked it. Any chance anyone has firmware I can apply?

witnessmenow, 2024-01-24:
Its very very tough to "brick" an esp32 device @jaymcd , you've just installed software on it that says the display is on the wrong pins. You'll have to program it with something that is intended for the board you have

mvsteen, 2024-03-31:
I am having  a hard time figuring out how to interface with this thing.
It's definitely a capacitive touch version, as there was a touch demo running on the device when i got it. Yet at the back it is labeled ESP32-2432S024. It measures 2.8 inches if i include the screen bezels and 2.4 if I don't.

So i am basically lost.

So if you have certain versions, you need to load different display libraries for getting it to work? I tried some example programs and they seem to upload but the display stays black.

rcarmo, 2024-08-12:
Is any of those ports able to act as a USB host?

clach04, 2025-02-09:
> Everyone referes to it as CYD2USB

I've been using CYD2 in my notes, CYD2USB works for me.

@witnessmenow how did you decide on this? Also thanks for having this repo/space!