witnessmenow/ESP32-Cheap-Yellow-Display #44: 3D Case not right

cymplecy, 2023-11-08:
The 3rd one in the list
https://thangs.com/designer/dimosram/3d-model/ESP32-2432S028%20enclosure%20for%20Battery%20and%20BMS-788656
is IMO completely wrong for the board

https://www.thingiverse.com/thing:5990927 is correct (but just having to reprint it 1mm bigger as VERY tight fit for me :) )

witnessmenow, 2023-11-18:
Added note, will leave this open cause maybe the whole list could be converted to a table

cymplecy, 2023-11-18:
There's been a mis-communication
the Thangs one is plain wrong 
[image]

[image]

No where to plug in the USB lead for instance :)  

The other one was just a bit tight for me and my printer but that was just social comment really :)

cymplecy, 2023-11-18:
Ignore my stuff about the Thang one :)

TheNitek, 2024-05-17:
@cymplecy so this means the case from thangs.com fits and is NOT "completely wrong"?

cymplecy, 2024-05-18:
I can't remember why I wrote my first post or why I subsequently said ignore me :)