witnessmenow/ESP32-Cheap-Yellow-Display #47: Need a solution/method for battery powering CYDs

witnessmenow, 2023-11-09:
It would be great to create a method or documentation around battery powering the CYD.

I'm sure it's very achievable using a cheap lipo IC like the [TP4056*](https://s.click.aliexpress.com/e/_DEaupN1) and possibly using the JST connector beside the USB, but a documented path to this would be great.

Not my area of expertise at all, but I do have some of the ICs ordered so I'll probably throw a stab at it shortly. But would be very happy for others input.

Also, maybe a stretch goal, but maybe even a PCB design specifically for the CYD , so it could be neatly mounted, might be cool

*Affiliate

suchmememanyskill, 2023-11-14:
I can confirm using the JST connector next to the usb port, you can provide 5V to VIN and ground to ground, and the display powers on. (using the model that also has a usb-c port)

I did notice some weird behaviour (web requests failing) while i was doing this, but i'm unsure if that's due to the way i'm powering the display or if my power source is just bad

bjinse, 2023-11-30:
You might check out this case, mentioned on the 3d models page, that links to a BMS and batteries.
https://thangs.com/designer/dimosram/3d-model/ESP32-2432S028%20enclosure%20for%20Battery%20and%20BMS-788656

BludzUK, 2024-03-14:
Hello all.
I'm new to world of programming and have been thankful for this GitHub page, it has allowed me to make a good start on a project I'm working on.
It wasn't until I had got quite far into the project that I realised that powering the CYD is going to be a bit of an issue, and then found that I wasn't alone.
Today I managed to find a workable solution. I tried a couple of different power modules from Amazon and found one that allows you to connect a 1s battery (18650 or lipo) and it ups the voltage to 5v (you adjust the output voltage with the dial). You can charge the battery using the usb c and the lights indicate when it is charging or charged.
Here's a link to the module, https://www.amazon.co.uk/gp/product/B0C3V847G2
I'll be working on a case in the near future that will house these extra components and maybe some physical buttons depending on how my project goes.
Hope this helps.
[image]
[image]

kf106, 2024-03-26:
Why is the switch needed? Is it just there for a manual power on/off or do you need to turn off the board when charging the battery? Can it be left out?

gjashley, 2024-03-26:
TP5046's are pretty safe. 
I used one of these boards: 
https://a.aliexpress.com/_mKCGlnc. It switches to the battery when it's unplugged (which also reboots the CYD so it's not really a UPS solution)
I have measured the power draw and it does drop way down to battery maintenance levels when all the power leds on the board are lit (ie when it measures 4.2v) 
There is a button which powers it off when it is running on battery.

BludzUK, 2024-03-26:
@kf106 It there was not a switch the unit just powers up when the battery is connected. Sadly the little charge/voltage regulator doesn't have a switch. I did also wonder if you could run the screen while charging so I wanted to test that. I can confirm that it works like I hoped and you can charge the battery while the screen remain running.
Honestly though I realise that a lot of the work I'm doing to make this screen work (not to mention all the work everyone else has done), is all to make the shit version of this screen/esp32 work. They sell a MUCH better version 3.2 inches with capactive touch instead of crap resistive touch, oh and it is IPS so the colours look way better, black actually is black rather than grey. They also added battery input direct on the board and it uses a single usb c to do charging and programing. Seems pointless wasting time making the shit version work when the best version is so cheap still (like $7).

BludzUK, 2024-03-26:
@gjashley Yea, I have some of those too actually, but it doesn't have a usb c for charging, so I assume you just add one?

gjashley, 2024-03-26:
Yeah, here's a project I've been working on with i2c sensors: [image]

alextipos, 2024-03-28:
> Yeah, here's a project I've been working on with i2c sensors: 

Sorry for the OT but can you please tell me where did you buy the JST connector and the measures? I'm lost, thanks a lot.

gjashley, 2024-03-28:
They're Micro JST 1.25MM 4-Pin Cables, I got mine from AliExpress, but you can probably get them faster from local suppliers: https://a.aliexpress.com/_mqkq4l8.
Make sure you get the 4-pin Male ones if you want pigtails which fit the port.

kf106, 2024-05-15:
I ordered a bunch of "5V 1A TYPE-C Micro USB 18650 TC4056A Lithium Battery Charging Board Charger Module with Protection Dual Functions TP4056" boards that were 1-50€ each. Scavenged a battery from a broken PS4 dualshock, and a broken bluetooth speaker provided both a connector for the battery, and a plug for the connector next to the micro USB port.

Five minutes of soldering, and it's charging the battery and powering the CYD. Unplugging the USB-C cable results in the display dimming ever so slightly, but the software keeps running.

It's a cheap solution, and it works. It's a shame they didn't include a TP4056 on the board and a battery connector like some of the boards out there.

[image]

lcorsini, 2024-07-19:
Is there an alternative way to power the board with a battery without using the serial jst connector?
because I have a gps receiver already connected to it
I thought of desolder the micro usb (mine has usb-c too) and connect the TP5046 and battery to that connector, is there a cleaner solution without removing the port? (apart from making a micro usb power cable)

SimpleAsh, 2024-07-28:
Hi I new to commenting, here but I read through info here and https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/discussions/87
And though I would have a try,
It seems to work, if anyone is interested 
https://youtu.be/XBvBRjAgCMo?si=fAkkX4ocsOttr9-3
[image]

lcorsini, 2024-07-28:
Yup, I know this works, but I'm already using the serial port for a gps
antenna

SimpleAsh, 2024-07-28:
Arr .. I see, I haven't looked at the pin out code yet, just looked at the board, I'll get ATGM336H NEO-6M on order for a play,

lcorsini, 2024-07-28:
Yeah I had one lying around and is working fine, but it occupies the serial connector that everyone uses for the battery

polarkernel, 2024-08-03:
> Yeah I had one lying around and is working fine, but it occupies the serial connector that everyone uses for the battery

@lcorsini Very interesting! I made a working solution for a NEO-6M using the GPIO pins 22 and 27 of the connectors P3 and CN1, but it is very annoying to have to use two JST's. I tried to use the RX and TX of connector P1, but the program hangs, when I try to connect to the serial port. Are these GPIO pins 1 and 3 on P1? Did you need any special setup?

Sorry to be a bit off-topic.

BludzUK, 2024-08-05:
@Vendetti 
Looks the same as the rest at a glance, but I started using these instead now. 
https://www.aliexpress.com/item/1005005138982767.html

VuDucVN-tech12, 2024-08-27:
> Hello all. I'm new to world of programming and have been thankful for this GitHub page, it has allowed me to make a good start on a project I'm working on. It wasn't until I had got quite far into the project that I realised that powering the CYD is going to be a bit of an issue, and then found that I wasn't alone. Today I managed to find a workable solution. I tried a couple of different power modules from Amazon and found one that allows you to connect a 1s battery (18650 or lipo) and it ups the voltage to 5v (you adjust the output voltage with the dial). You can charge the battery using the usb c and the lights indicate when it is charging or charged. Here's a link to the module, https://www.amazon.co.uk/gp/product/B0C3V847G2 I'll be working on a case in the near future that will house these extra components and maybe some physical buttons depending on how my project goes. Hope this helps. ![IMG_20240314_181059405](https://private-user-images.githubusercontent.com/13721144/312953982-4f4c13f9-8414-42a3-94c3-fa8f8f51d7a9.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ3MzMxOTUsIm5iZiI6MTcyNDczMjg5NSwicGF0aCI6Ii8xMzcyMTE0NC8zMTI5NTM5ODItNGY0YzEzZjktODQxNC00MmEzLTk0YzMtZmE4ZjhmNTFkN2E5LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI3VDA0MjgxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU3YmRiOWNmMDY1MjNiY2YyNjEyOTY4MjAyY2YwNTQ2MmEzYWI2ZDk2Yjk2ZjgzN2U2ZGU3ZDM0Nzk0ODg5ZDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.R695aceZDhPOyarGD4Rgk3NUKgmCd3CMbwA8-F8kr-k) ![IMG_20240314_181122799](https://private-user-images.githubusercontent.com/13721144/312953993-75d68fbd-db3b-4099-8251-544018e0383e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ3MzMxOTUsIm5iZiI6MTcyNDczMjg5NSwicGF0aCI6Ii8xMzcyMTE0NC8zMTI5NTM5OTMtNzVkNjhmYmQtZGIzYi00MDk5LTgyNTEtNTQ0MDE4ZTAzODNlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI3VDA0MjgxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcyMzg5ZDJiOTlkZWU5OGY2MDA4N2I1ZDljYTVhYTY0OWIwMzhkMjJkMTQ3ZTFkZGI2YmRmOWIzMmFjMGZmZmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.y1Hapb5xqTCVJc_vlJ7RTwzqZpEPbdQPK0OGqmj0Sls)

Hey man! Could you please take a look and let me know if my connections are correct and if this setup will work properly like yours? I'd greatly appreciate any feedback or suggestions you might have to improve it. I've attached an image of my schematic for your reference.
[image]

SimpleAsh, 2024-08-27:
Give me a sec module is not fantastic and switch is not needed but I'll get my PC out and hev a look, your

SimpleAsh, 2024-08-27:
[image]

SimpleAsh, 2024-08-27:
[image]

SimpleAsh, 2024-08-27:
The correct boards help

SimpleAsh, 2024-08-27:
I'm not a master so..

SimpleAsh, 2024-08-27:
One board and battery was all that was needed different board 🛹 might be harder to turn 😂

VuDucVN-tech12, 2024-09-18:
Im using tp4056 charging module with the mini 3.7v to 5v step up converter and connect with the JST 4-pin next to the USB. it can power it, but when i do any kinds of attacks (e.g Sour Apple, Evil Portal ) it will reboot the CYD ? Any solution ?

AllanOricil, 2024-12-25:
Is there an issue to use a power bank?

jayb880, 2025-01-10:
there is a way of connecting a battery and c type charger without using any of the JST connections