witnessmenow/ESP32-Cheap-Yellow-Display #48: Core Dump!

dkim777, 2023-11-09:
Got excited about this project and ordered two.
Tried the first couple of examples but got following error.
any ideas?   both are doing same thing on all two of my devices.
Thanks

-------------------------------
E (157) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
E (134) spi_flash: Detected size(4096k) smaller than the size in the binary image header(16384k). Probe failed.

assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)

dkim777, 2023-11-10:
Changed the Flash mode to DIO from QIO, and now I get different error.
-----------------------------------

ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
E (8) gpio: gpio_set_level(227): GPIO output gpio_num error

witnessmenow, 2023-11-18:
Not sure what is up with this. Have you gone through the [Troubleshooting](https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/blob/main/TROUBLESHOOTING.md) steps?

dkim777, 2023-11-21:
Thanks for the troubleshooting step.
You might want to add following.
---------------
You may need to put the device into download mode by press and hold "BOOT" and click "RESET" to change mode to download and then press and hold when you see "Connecting ........"  after the compile to properly upload the code.
---------------
not sure why but above method works, it won't just flash like other esp32 devices.

closed 2023-11-21 by dkim777