witnessmenow/ESP32-Cheap-Yellow-Display #51: ESP32-2432s028r Freetouchdeck

nikvs216, 2023-11-14:
Hi, i tried to run Freetouchdeck on from a fork from Raycast7373 but i could not get it to work, i contacted him about is but he does not own a  ESP32-2432s028r, and could not help me much, Does anyone maybe got Freetouchdeck running on  ESP32-2432s028r,
These are my pins in User_setup.h
#define TFT_MISO 12
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
//#define TFT_RST   4  // Reset pin (could connect to RST pin)
#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
#define TOUCH_CS 33

witnessmenow, 2023-11-18:
I know Dustin was doing some re-work on the FTD code to make it more adaptable to a display configured like the CYD, but im not sure where he has landed on it.

killabeenl, 2024-02-20:
What is the status in this issue?
I tried to flash a 2432s028r manually but can't get it to work.
Is it possible to add this board to the web install page?