witnessmenow/ESP32-Cheap-Yellow-Display #57: Visual Studio Error with Touch

scudrunner, 2023-11-16:
I am using Visual Studio for the CYD, and I have the screen working with the following build flags:
build_flags =
    -D USER_SETUP_LOADED
    -D ILI9341_2_DRIVER
    -D TFT_MISO=12
    -D TFT_MOSI=13
    -D TFT_SCLK=14
    -D TFT_CS=15
    -D TFT_DC=2
    -D TFT_RST=-1
    -D LOAD_GLCD=1
    -D LOAD_FONT2
    -D LOAD_FONT4
    -D LOAD_FONT6
    -D LOAD_FONT7
    -D LOAD_FONT8
    -D LOAD_GFXFF
    -D SMOOTH_FONT
    -D SPI_FREQUENCY=55000000
    -D SPI_READ_FREQUENCY=20000000
    -D SPI_TOUCH_FREQUENCY=2500000
    -D TFT_WIDTH=240
    -D TFT_HEIGHT=320
    -D TFT_BL=21            
    -D TFT_BACKLIGHT_ON=HIGH  

I did not see the touch_cs defined in the setup, and I am getting build errors trying to run the touch example in here.  I am not sure if the code is wrong, or if I am doing something wrong.  Build log portion is included below 

Dependency Graph
|-- TFT_eSPI @ 2.5.33
|-- XPT2046_Touchscreen @ 0.0.0-alpha+sha.26b691b2c8
|-- SPI @ 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
In file included from src/main.cpp:34:
.pio/libdeps/esp32dev/TFT_eSPI/TFT_eSPI.h:973:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
        ^~~~~~~
src/main.cpp: In function 'void setup()':
src/main.cpp:66:17: error: no matching function for call to 'XPT2046_Touchscreen::begin(SPIClass&)'
   ts.begin(mySpi);
                 ^
In file included from src/main.cpp:28:
.pio/libdeps/esp32dev/XPT2046_Touchscreen/XPT2046_Touchscreen.h:46:7: note: candidate: 'bool XPT2046_Touchscreen::begin()'
  bool begin();
       ^~~~~
.pio/libdeps/esp32dev/XPT2046_Touchscreen/XPT2046_Touchscreen.h:46:7: note:   candidate expects 0 arguments, 1 provided 
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1

scudrunner, 2023-11-16:
I think my error is most likely related to the touchscreen library
no matching function for call to 'XPT2046_Touchscreen::begin(SPIClass&)'

witnessmenow, 2023-11-16:
Someone on the discord had the same error, point to the GitHub of the touch library

scudrunner, 2023-11-16:
I found the error that witnessmenow mentioned - the platfrom IO library handler does not grab the current version, even when I pointed directly to the git.  I had to fork the library, release it, and then point to my release...

closed 2023-11-16 by scudrunner

mart-e, 2023-11-18:
@scudrunner how did you fixed your issue? I have the same error and using platform IO CLI. Thanks

scudrunner, 2023-11-18:
I gave up on platform IO.  It was not loading libraries correctly and was
giving me to many issues.  I was forking libraries and publishing then just
to load the right files….

mart-e, 2023-11-19:
Ok I see, `pio pkg install -l XPT2046_Touchscreen` installs version `0.0.0-alpha+sha.26b691b2c8` from [registry.plateformio.org](https://registry.platformio.org/libraries/paulstoffregen/XPT2046_Touchscreen) (2019) and not the one [from github](https://github.com/PaulStoffregen/XPT2046_Touchscreen) (1.4).

Installing using `pio pkg install -l "https://github.com/PaulStoffregen/XPT2046_Touchscreen.git"` does seems to take the latest version (`XPT2046_Touchscreen@0.0.0+sha.5d5120e` which is [the latest commit](https://github.com/scudrunner/XPT2046_Touchscreen/commit/5d5120e) as of today) and I can run `pio run` correctly now.

scudrunner, 2023-11-19:
> take the latest version ***@***.***+sha.5d5120e which is the

I was trying to run some other packages as well, and had the same issue, so
I gave up.  If it was only this package, I would have been OK, but it was
to much to deal with.