witnessmenow/ESP32-Cheap-Yellow-Display #76: Wrong colors

hello-world-dot-c, 2023-11-29:
I've just got myself one CYD and getting started. I went through the set up for lcgl (lv_conf.h) and TFT_eSPI (User_Setup.h). I can build the LVGL example and in principle, everything works but the colors are all "negative", looking like this:

[image]

When I use tft.invertDisplay() it looks better but the colors are "washed out" , the red here is very pale and the green looks like a grey:

[image]

This is how the demo looked when I received the display:

[image]

Anybody knows how to fix this? LV_COLOR_16_SWAP 1 in lv_conf.h doesn't help and screws the colors up completely.

witnessmenow, 2023-11-29:
This is a known issue with the version with two USB ports.

You've already sorted the first issue of the inverted colours, there is some discussion about fixing the colours here:

https://github.com/Bodmer/TFT_eSPI/issues/2985

hello-world-dot-c, 2023-11-29:
Cool, that fixed it for me, thanks!

closed 2023-11-29 by hello-world-dot-c