witnessmenow/ESP32-Cheap-Yellow-Display #108: blank screen for 10slideshow example

matroll, 2024-01-03:
hey, thanks a ton for your work on this H/W! looks amazing. just got it and will try to make a miniTV.
unfortunately when i try to run the 1-slideshow sketch, i only get a blankscreen.
- SD is fine and it finds the file(s)
- jpgs are 320x240
- screen is fine (other ex work and i can do a fillscreen RED in the slideshow sketch - works fine)

i get the output below on the monitor but the screen stays black (or red if i do the test in red)

any idea?

thanks

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
FAT type:   FAT16
JPGs found: 6
Loading file no 0
File: seinfeld.jpg

witnessmenow, 2024-01-20:
Are you using webflash, platform io or Arduino ide? Have you tried the simpler examples?

matroll, 2024-01-21:
thanks for getting back to me. arduino ide  (latest version and libraries up to date) and yes the simple screen examples work good. sd is ok too as i can get the image name and size.

witnessmenow, 2024-01-21:
Was it a while ago you setup the user_setup for the display? About a month ago @TheNitek made a fix to the user_setup that might be the problem if you don't have it. So could you try update your user_setup with the latest and see does that work for you?

matroll, 2024-01-21:
away from home for a few days but will try that upon my return. indeed the user setup did not work and had to do my own derived from another one. thanks for the tip, looking forward to trying  it. best

matroll, 2024-01-24:
hello brian,

that was the issue, fantastic! thanks for the help!

so i looked at the edits, and it was only a matter of using HSPI?

 

now i'm on to trying to run some video in mjpg format. i'm going off of the Mini_TV <https://github.com/moononournation/MiniTV>  example.

 

i did notice that the displaying of the jpg is not "instant" (there is a quick but noticeable sweep down of the image). i hope it'll be quick enough to do a decent video frame rate. i tried setting the spi to 80mhz and the effect is still noticeable.

 

cheers, and thanks again for helping!

mat.

matroll, 2024-01-24:
brian,

 

sorry to bug you again but while running into problems with video, i re-watched your video and noticed that you were playing video on the CYD (muppets) >> could you share the code for that? i imagine it's some form of mjpeg as well?

 

thanks in advance.

mat.

cgreening, 2024-01-24:
I need to do some tweaks for the CYD - at the moment I can't get DMA to work on the display, but it's a simple fix (I think...).

matroll, 2024-01-24:
thanks both so much for your replies.

chris, please let me know if you find a fix, i'd be interested in running it.

 

in the meantime, i managed to get some mjpg runnning with the miniTV examples but i get some minimal banding that i'm trying to solve (i think it's a question of buffer sizes)

 

thanks to both of you.