witnessmenow/ESP32-Cheap-Yellow-Display #115: 3D Stand for CYD request

Grey-Lancaster, 2024-01-17:
@88markus88 
Seems like the only way to contact you is to raise and issue. 

This fits perfectly for the new USB C board, but I prefer a stand like this:
https://www.thingiverse.com/thing:4768766

it would require a 90 degree adapter like this
https://www.amazon.com/AreMe-Extender-Compatible-Notebook-Computers/dp/B0B2NJ3P3L

Think you could make that?
Thanks,
Grey

88markus88, 2024-01-18:
Hi Grey,

I just looked at the stand you pointed to - cool idea! However, that is a
completely new design, not a modification of the housing that I have made.
You cannot just add feet to the main body of the housing, that would make
it difficult to print (lots of overhangs).

I will look into it, and then let you know. Probably on Thingiverse, since
I am having trouble to post changes here as Pull Requests ;-)

Best, Markus

Grey-Lancaster, 2024-01-18:
Great
I was thinking about modifying the top.  add an angle foot and printed facedown like the thinverse link. Just leave the back open.  But the holes would have to be reconfigured to take the inserts. Probably need to be a bit longer to allow for the 90  degree adapter
If you make it drop a note here please.
Grey

88markus88, 2024-01-19:
Hi Grey,
right, the display cannot be attached to the front. 
I've just remembered hat I did some feet that can be clipped onto the display in both landscape and portrait mode. I've put the whole thing on [Thingiverse](https://www.thingiverse.com/thing:6440252). Please have a look and tell me if this fits your needs. I will probably make a version with included stand anyway, but that will certainly take a while.
Markus

Mordevolt, 2024-01-20:
just one more 3d printed model:
https://www.printables.com/model/645166-esp32-28inch-touch-ips-cyd-lcd-screen-case