witnessmenow/ESP32-Cheap-Yellow-Display #120: rolling clock project -- days of week -- longer days not cleared in display eg one gets  "SSundayy"

danpeirce, 2024-01-23:
When day shifts from Saturday to Sunday it displays as "SSundayy"
Could possibly pad shorter day names with spaces or clear that part of display when rolling over to a new day.

https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/blob/main/Examples/Projects/RollingClock/RollingClock.ino

witnessmenow, 2024-01-23:
Are you using the latest code? @hwiguna submitted a fix for this a few weeks ago

https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/pull/104

danpeirce, 2024-01-23:
Oh. Sorry. Looks like you are correct. I pulled the change but failed to reprogram my cyd. Looks like that fix should work. I won't see it now until Thursday but I should close this.