witnessmenow/ESP32-Cheap-Yellow-Display #127: ESP32-2432s022 ESPHome

dirtyharryiv, 2024-01-27:
Does somebody have this version and got it running with esphome? I only have a white screen with the provided examples.
The display is a little bit smaller but it other than it looks like a CYD

mstewart197930, 2024-02-07:
I have a similar issue on a different project, it would seem these are a different display chip and pin out, to the slightly bigger ones.

guybw, 2024-03-21:
Just to add, I have the same issue.

nielsnl68, 2024-04-23:
Please ping me on the esphome discord and i will try to see what i can do for you.

nielsnl68, 2024-04-23:
Okay, it looks that they are using the Parallel interface (i80) between display and esp32.
At this moment in time there is no support in esphome yet. We are working to get this implemented very soon.