witnessmenow/ESP32-Cheap-Yellow-Display #128: HelloRadio example not compiling

Grey-Lancaster, 2024-01-28:
No clue unless the latest audio.h broke it?

Arduino: 1.8.16 (Windows 10), Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

In file included from F:\Users\greyl\Documents\Arduino\CYD_Radio\CYD_Radio.ino:46:

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:35:50: error: variable or field 'audio_id3image' declared void

 extern __attribute__((weak)) void audio_id3image(File& file, const size_t pos, const size_t size); //ID3 metadata image

                                                  ^~~~

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:35:50: error: 'File' was not declared in this scope

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:35:50: note: suggested alternative:

In file included from F:\Users\greyl\Documents\Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_ESP32.h:138,

                 from F:\Users\greyl\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:101,

                 from F:\Users\greyl\Documents\Arduino\CYD_Radio\CYD_Radio.ino:40:

C:\Users\greyl\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\FS\src/FS.h:47:7: note:   'fs::File'

 class File : public Stream

       ^~~~

In file included from F:\Users\greyl\Documents\Arduino\CYD_Radio\CYD_Radio.ino:46:

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:35:56: error: 'file' was not declared in this scope

 extern __attribute__((weak)) void audio_id3image(File& file, const size_t pos, const size_t size); //ID3 metadata image

                                                        ^~~~

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:35:56: note: suggested alternative: 'pipe'

 extern __attribute__((weak)) void audio_id3image(File& file, const size_t pos, const size_t size); //ID3 metadata image

                                                        ^~~~

                                                        pipe

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:35:62: error: expected primary-expression before 'const'

 extern __attribute__((weak)) void audio_id3image(File& file, const size_t pos, const size_t size); //ID3 metadata image

                                                              ^~~~~

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:35:80: error: expected primary-expression before 'const'

 extern __attribute__((weak)) void audio_id3image(File& file, const size_t pos, const size_t size); //ID3 metadata image

                                                                                ^~~~~

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:36:51: error: variable or field 'audio_id3lyrics' declared void

 extern __attribute__((weak)) void audio_id3lyrics(File& file, const size_t pos, const size_t size); //ID3 metadata lyrics

                                                   ^~~~

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:36:51: error: 'File' was not declared in this scope

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:36:51: note: suggested alternative:

In file included from F:\Users\greyl\Documents\Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_ESP32.h:138,

                 from F:\Users\greyl\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:101,

                 from F:\Users\greyl\Documents\Arduino\CYD_Radio\CYD_Radio.ino:40:

C:\Users\greyl\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\FS\src/FS.h:47:7: note:   'fs::File'

 class File : public Stream

       ^~~~

In file included from F:\Users\greyl\Documents\Arduino\CYD_Radio\CYD_Radio.ino:46:

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:36:57: error: 'file' was not declared in this scope

 extern __attribute__((weak)) void audio_id3lyrics(File& file, const size_t pos, const size_t size); //ID3 metadata lyrics

                                                         ^~~~

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:36:57: note: suggested alternative: 'pipe'

 extern __attribute__((weak)) void audio_id3lyrics(File& file, const size_t pos, const size_t size); //ID3 metadata lyrics

                                                         ^~~~

                                                         pipe

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:36:63: error: expected primary-expression before 'const'

 extern __attribute__((weak)) void audio_id3lyrics(File& file, const size_t pos, const size_t size); //ID3 metadata lyrics

                                                               ^~~~~

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:36:81: error: expected primary-expression before 'const'

 extern __attribute__((weak)) void audio_id3lyrics(File& file, const size_t pos, const size_t size); //ID3 metadata lyrics

                                                                                 ^~~~~

In file included from F:\Users\greyl\Documents\Arduino\CYD_Radio\CYD_Radio.ino:46:

F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:448:5: error: 'File' does not name a type; did you mean 'pipe'?

     File                  audiofile;    // @suppress("Abstract class cannot be instantiated")

     ^~~~

     pipe

Multiple libraries were found for "Audio.h"

 Used: F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S-master

 Not used: F:\Users\greyl\Documents\Arduino\libraries\ESP32-audioI2S

exit status 1

Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

CoolmdXi, 2024-02-13:
I have almost exactly the same issues

matthew1001, 2024-03-09:
I cloned the `3.0.0` release and imported that into the IDE to get round this (`git clone https://github.com/schreibfaul1/ESP32-audioI2S.git --branch 3.0.0`)