witnessmenow/ESP32-Cheap-Yellow-Display #130: Everything needs to be updated

Grey-Lancaster, 2024-01-29:
This may have been great once upon a time, but examples no longer compile in Arduino or Paltfromio.

Problem with Platformio is spiffs is no longer supported.

Can anyone show me how to do the [slideshow example](https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/tree/main/Examples/Showcases/1-Slideshow) for example with current library's?

Thanks,
Grey

Grey-Lancaster, 2024-01-29:
for example the [LVGL demo](https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/tree/main/Examples/LVGL) does not work with the latest and greatest LVGL library v(.0 it has to be v8.3.9

Grey-Lancaster, 2024-01-29:
I cannot seem to find what you must be talking about.  I just looked and apparently the was or is a standalone Platformio IDE.  But whenever I click on download IDE it goes to Platformio for VS code which is  what I have and SPIFFS is no longer supported.

https://community.platformio.org/t/problem-with-spiffs/36837

[image]

witnessmenow, 2024-01-29:
> I cannot seem to find what you must be talking about. I just looked and apparently the was or is a standalone Platformio IDE. But whenever I click on download IDE it goes to Platformio for VS code [...]

That looks to relate to esp8266 not esp32

Grey-Lancaster, 2024-01-29:
That looks to relate to esp8266 not esp32
You might be right, but it still gives an error. Are you using VS Code with platformio ?
 In his examples he uses littlefs
https://registry.platformio.org/libraries/bodmer/TFT_eSPI/examples/PNG_Images

When I build it does not include spiffs

Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TFT_eSPI @ 2.5.43
|-- JPEGDEC @ 1.4.0
|-- SdFat @ 2.2.2
|-- XPT2046_Bitbang_Slim @ 2.0.0
|-- SPI @ 2.0.0
|-- FS @ 2.0.0
|-- LittleFS @ 2.0.0
Building in release mode
Compiling .pio\build\cyd2usb\src\src\CYC_Slideshow.cpp.o

Successfully created esp32 image.
Compiling .pio\build\cyd2usb\lib4c1\FS\FS.cpp.o
Compiling .pio\build\cyd2usb\lib4c1\FS\vfs_api.cpp.o
In file included from .pio/libdeps/cyd2usb/TFT_eSPI/TFT_eSPI.h:101,
                 from src/CYC_Slideshow.cpp:14:
.pio/libdeps/cyd2usb/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:139:12: fatal error: SPIFFS.h: No such file or directory

****************************************************************
* Looking for SPIFFS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPIFFS.h"
* Web  > https://registry.platformio.org/search?q=header:SPIFFS.h
*
****************************************************************

Grey-Lancaster, 2024-01-30:
1000 pardons
SumB*tch
https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/tree/main/Examples/Showcases/1-Slideshow
The Arduino code works GREAT if you follow directions
I installed XPT2046_Bitbang which did not work.
XPT2046_Bitbang_Slim (SLIM) works great. Touch moves to next picture or the delay does.
Love it !!
Platformio VS code still fails, but who cares if Arduino works,

closed 2024-01-30 by Grey-Lancaster