witnessmenow/ESP32-Cheap-Yellow-Display #137: Add CYD Variants list

aamott, 2024-02-17:
I'd like to request a place to list the variants of CYD. I recognize that this is built around the micro-USB-only variant, but the USB-C variant is common too, end I can get it shipped for $9.50. Works perfectly with CYD-Klipper as long as I invert the colors (a fortunate feature) and I can imagine lots of people buying that variant without knowing better as I did. Galago works on it too.
There are other variants too, and I think with some basic documentation, we could take advantage of a lot of these with minimal or no changes.

As a start, the USB-C version has a slightly worse screen (more gray blacks) but comes cheaper. It's screen is inverted according to the CYD-Klipper developer, but otherwise it's the same. 

There's also at least one larger version of the screen, but I've only heard of a few people trying it. One has higher resolution.

witnessmenow, 2024-02-18:
The USB C version is supported as a cyd per these instructions

https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/blob/main/cyd.md

All other of these yellow displays are not considered CYDs, but I have already added a place to add info about them here

https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/tree/main/Variants

william-orange, 2024-02-19:
Does anyone have a schematic for the CYD2USB board?

A couple of notes.  The gamma correction mentioned in the link above makes an enormous difference.  My board programs fine through a cable like this _USB 10Gbps  Cable A Male to C Male 3.00'_

aly-fly, 2024-04-14:
I agree, schematic or pinout would be very welcome.
Also I wasn't able to find a table that lists LCD driver chip types for every variant.

aly-fly, 2024-04-14:
One staring point for the board info is attached. Translated from Japanese.
[ESP32-2432S028_macsbug.pdf](https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/files/14969885/ESP32-2432S028_macsbug.pdf)

Grey-Lancaster, 2024-09-26:
> The gamma correction mentioned in the link above makes an enormous difference.

Where did you insert that code in .ino?

Grey-Lancaster, 2024-09-27:
I got it to build on the cyd2 but the colors are still off.  I swear it used to work fine

```
  digitalWrite(16, HIGH);
  digitalWrite(17, HIGH);

  // Start the tft display and set it to black
  tft.init();
  tft.setRotation(1); //This is the display in landscape

  // Define the ILI9341_GAMMASET if it is not included in your library
#define ILI9341_GAMMASET 0x26  // Gamma curve command for ILI9341

    // Insert the gamma curve setup here
  tft.writecommand(ILI9341_GAMMASET); // Gamma curve selected
  tft.writedata(2);
  delay(120);
  tft.writecommand(ILI9341_GAMMASET); // Gamma curve selected
  tft.writedata(1);
  
  // Clear the screen before writing to it
  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.setTextSize(2);
  tft.setSwapBytes(true);

  ts.begin();
```

william-orange, 2024-09-27:
In setup() {
 
// lots of stuff and toward the end
if (CYD2USB) gammaCorrection();  // improves color on CYD2USB
}
Here's the function:

/*---------------------------------------------- Gamma correction  --------------------------------------------------------------*/

void gammaCorrection() {
  tft.writecommand(ILI9341_GAMMASET);  //Gamma curve selected
  tft.writedata(2);
  delay(120);
  tft.writecommand(ILI9341_GAMMASET);  //Gamma curve selected
  tft.writedata(1);
}

So, your code looks fine.

Bill

Grey-Lancaster, 2024-09-27:
Thanks, I finally got it working even though it is usbc but I had to change some other stuff. it seems to work with the 9341-2 driver

so I added a section to platformio.ini

```
[env:cyd]
build_flags =
	${env.build_flags}
	-DILI9341_2_DRIVER
	

[env:cyd2usb]
build_flags =
	${env.build_flags}
	-DST7789_DRIVER
	-DTFT_INVERSION_OFF

[env:cyd2b]
build_flags =
	${env.build_flags}
	-DILI9341_2_DRIVER
	-DTFT_INVERSION_ON
	-DENV_CYD2B
	-DUSE_GAMMA_CORRECTION

```
then in .ino

```
  // Start the tft display and set it to black
  tft.init();
  tft.setRotation(1); //This is the display in landscape

 // Define the ILI9341_GAMMASET if it is not included in your library
#define ILI9341_GAMMASET 0x26  // Gamma curve command for ILI9341
#ifdef ENV_CYD2B
  // Code specific to cyd2b
    // gamma curve setup 
  tft.writecommand(ILI9341_GAMMASET); // Gamma curve selected
  tft.writedata(2);
  delay(120);
  tft.writecommand(ILI9341_GAMMASET); // Gamma curve selected
  tft.writedata(1);  
#endif

  // Clear the screen before writing to it
  tft.fillScreen(TFT_BLACK);

```

PRSXFENG, 2024-10-16:
> I'd like to request a place to list the variants of CYD.

There is this repo that lists a lot of CYD and CYD adjacent boards
https://github.com/rzeldent/platformio-espressif32-sunton

for me personally the 4 I see the most common are
MicroUSB only, 2USB, 2USB with ST7789 display, and USB-C only 2.4inch variant

aamott, 2024-10-16:
Now that's a list! Would be sweet if the most popular ones could be added here with notes on the caveats or differences, since that one specifically shows technical info. Otherwise, I'm fine with closing this.