witnessmenow/ESP32-Cheap-Yellow-Display #213: TOUCH_CS undefined warning in hello world example

mcl-uk, 2024-06-18:
I'm a complete newbee to CYD - thanks so much for your brilliant intro into using this kit.
Trying the Hello World example I was getting TOUCH_CS undefined warnings, and a blank LCD screen.
Uncommenting one of the TOUCH_CS lines in User_Setup.h got it working.
But I'm unsure what the correct setting for TOUCH_CS should be, your Pin Descriptions page indicates it might be 33, is this right?
If so it might be worth updating the User_Setup.h file to include this and avoid confusion for other newbees like me.
Thanks again.
Steve.

TheNitek, 2024-06-18:
The touch screen is not sharing a bus with the display (see https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/blob/main/PINS.md#touch-screen ), which is not supported by the TFT_eSPI library. The warning is nothing to worry about. For details on how to setup the thouch screen you can have a look at the other examples.

closed 2024-06-18 by TheNitek

TheNitek, 2024-06-18:
There should be no correlation between that warning an the screen working (/not working)

mcl-uk, 2024-06-18:
Quite! But that was what happened.

mcl-uk, 2024-06-18:
I'm using arduino IDE.

Zynth9999, 2024-06-19:
Ah then start using PIO, it worked for me and it's overall better.

pedz, 2025-06-13:
Hi... I am using PlatformIO which I am guessing what you call PIO.  And I have tested the 2-TouchTest example and it works but I'm confused what the warning message from TFT_eSPI.h is about.  What are they trying to warn me about?  (I'm mostly just curious... warnings that are ignored tend to come bite me later on.)

Zynth9999, 2025-06-14:
Im sorry, i can't check right now. I usually ignore warnings