witnessmenow/ESP32-Cheap-Yellow-Display #216: Compile error in Basics/HelloRadio.ino example

mcl-uk, 2024-06-19:
I'm getting an "I2S_DAC_CHANNEL_LEFT_EN undefined" error when comiling this example sketch. I tried #defining this to 2 (found a reference to it in a legacy h file) and now the sketch compiles and runs seemingly correctly but I get no audio out from the CYD's spkr port. Has anyone found a way to get this example to work?
Thanks for reading this far.

paelzer, 2024-06-19:
Hello

Guess pin 2 is wrong. In the pin description GPIO 26 is mentioned to be the correct pin for I2S_DAC_CHANNEL_LEFT_EN.

https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/blob/main/PINS.md

Boris

mcl-uk, 2024-06-20:
Thanks for your response. Sadly this doesn't work, still no signal at the spkr pins. If I add the line "i2s_set_dac_mode(I2S_DAC_CHANNEL_LEFT_EN);" to the code I get an undefined error for i2s_set_dac_mode.
Be grateful for any other ideas.

TheNitek, 2024-06-24:
Memo to myself: https://discord.com/channels/630078152038809611/637812536158453780/1254755798727266417