witnessmenow/ESP32-Cheap-Yellow-Display #224: ESP32-3248S035

Patsjemoe, 2024-06-29:
Hi,
I ordered ESP32-3248S035C but  received the ESP32-3248S035 without suffix.
I is impossible to get information from the supplier on wether this one has a touch screen, and if it is R or C.
Anyone had the same experience ?
Anyone knows what this part is ?
thanks in advance
[image]

TheNitek, 2024-06-29:
Power it on and see if the touchscreen works with the preinstalled demo

Patsjemoe, 2024-06-29:
Hi,
Thank you for your quick feedback.
That was a screen with a call girls phone number, and is already
overwritten with tests I am doing.
 I am trying some other examples, but screen does not responds. Not sure if
I adapted well enough some s028 examples to s035.
Do you know of a working example for 3248S035?
Regards
Ludo

mastertape, 2024-08-30:
did it work?