witnessmenow/ESP32-Cheap-Yellow-Display #233: ESPHome 6-VariousFeatures: "Pin 25 is used in multiple places"

littleyoda, 2024-07-13:
https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/tree/main/Examples/ESPHome/6-VariousFeatures

I'm unable to compile the ESPHome example.
The Pin 25 is used in multiple places. (spi_touch/clk_pin and i2s_audio/i2s_bclk_pin)

 ```
spi:
  - id: spi_tft
    clk_pin: GPIO14
    mosi_pin: GPIO13
    miso_pin: GPIO12
  - id: spi_touch
    clk_pin: GPIO25
    mosi_pin: GPIO32
    miso_pin: GPIO39

```

```
i2s_audio:
  i2s_lrclk_pin: GPIO25
  i2s_bclk_pin: GPIO26

```

```
Failed config

spi: [source yellowtft1.yaml:176]
  
  Pin 25 is used in multiple places.
  - id: spi_tft
    clk_pin: 
      number: 14
      mode: 
        output: True
        input: False
        open_drain: False
        pullup: False
        pulldown: False
      inverted: False
      ignore_pin_validation_error: False
      ignore_strapping_warning: False
i2s_audio: [source yellowtft1.yaml:212]
  
  Pin 25 is used in multiple places.
  - i2s_lrclk_pin: 25
    i2s_bclk_pin: 26

```

TitoTB, 2024-07-22:
Same issue here!

littleyoda, 2024-07-22:
with the following change you can at least compile it. 
However, I have not yet checked whether sharing the PIN makes sense.

```
spi:
  - id: spi_tft
    clk_pin: GPIO14
    mosi_pin: GPIO13
    miso_pin: GPIO12
  - id: spi_touch
    clk_pin: 
      number: GPIO25
      allow_other_uses: true
    mosi_pin: GPIO32
    miso_pin: GPIO39
```

````

i2s_audio:
  i2s_lrclk_pin:
      number: GPIO25
      allow_other_uses: true
  i2s_bclk_pin: GPIO26
```

TitoTB, 2024-07-22:
Thanks a lot!

I replace GPIO25 with GPIO27 and i can compile the code also.

However i just hear noise through the speaker. I don't know if it is the yaml configuration, the speaker, or the GPIO... any help?

TitoTB, 2024-07-24:
I tried both ways with no luck. I also tried with different speaker.

Has anyone achieve to connect and use a speaker with esphome?