witnessmenow/ESP32-Cheap-Yellow-Display #261: How to use W25q32 (U4) external Flash?

TheSnowFall, 2024-10-07:
Hi,
I have a CYD  2432S028 display. At the U4, I am planning to use a flash chip (W25q32) which is already routed to the SDSPI bus. I am a noob in using flash chips. Moreover, it is connected to the SD bus which I have never encountered. But I want to learn.
I do not know how to use it as :
1. As just an external memory device
2. Use as main flash instead of using 4Mb internal flash
3. How to use both memories(internal & external)

I generally use ESP-IDF to code ESP32. I also have good knowledge of arduino IDE. But I have never used platform.io. 

**Please enlighten me**. I added the screenshot of the portion of the schematic of W25q32 routed to the SDSPI bus in CYD
[image]

JMiahMan1, 2024-11-08:
Hello, I found this post looking to see if it was possible as I was wondering the same thing. Couldn't find any info on it so I just went for it not really having a clue what I was doing. I was just able to solder on a W25Q64FWSIG that I bought on Amazon (https://www.amazon.com/gp/product/B0B45NTQRL).

[image]

Pin 1 is on the bottom left match that up with the tiny dot on the chip (matching https://macsbug.wordpress.com/2022/08/17/esp32-2432s028/)

To get it to work I had to do two things.

1. Edit my partition table by creating a CSV file with this included
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x200000,
app1,     app,  ota_1,           ,0x200000,
spiffs,   data, spiffs,          ,0x1F0000,
```

2. I used these options to build in my YAML
```
esphome:
  platformio_options:
    board_build.f_flash: 80000000L
    board_build.flash_size: 8MB
    board_build.partitions: "/home/user/Desktop/ESP-Devices/CYD/YAML/part.csv"

esp32:
  board: esp32dev
  framework:
    type: esp-idf
  flash_size: 8MB
```
I flashed connected to my machine over USB A to C because I had previously edited my partition table to remove one of the OTA Partitions to give me more room. After that initial flash, I went back and tried a wireless OTA flash to make sure all was well as this is really the reason I did this. This config has 2 OTA 2MB partitions set, if you need more room feel free to adjust. Unfortunatly, I seem to be having an issue with over the air updates getting a "ERROR Error receiving acknowledge binary size: timed out" lately I have seen a lot of bug reports for this issue so I am not sure if it's related or not, in any case I am able to flash over USB for now, but not being able to flash OTA is kinda the whole reason I did this so I'm going to continue to troubleshoot that and see if it's related to this recent addition. Just want to give you what I have done so far in hopes it may help you or someone else.

Kind regards, 

Jeremiah

TheSnowFall, 2024-11-08:
Hi,
Thanks @JMiahMan1 .
I would try that also.

> "ERROR Error receiving acknowledge binary size: timed out"

Definitely, this is not coming for anything related to flash chip/OTA partition.

But I need a clarification. See them below:

> My machine over USB A to C 
what it means?

I will also do a deep dive into the OTA issue. Will update you.

JMiahMan1, 2024-11-10:
Just not using an all USB C cable, but a A to C
[image]

Some boards have issues because the resistors are missing on the control lines that USB-C uses for detection so using a A to C cable is the best way to work around that people run into issues with some boards using a C to C Cable.