witnessmenow/ESP32-Cheap-Yellow-Display #271: colors.

rocco99, 2024-11-17:
TFT_BLACK displays white, also cannot find a way to rotate text to be viewed in landscape orientation. Other colors also not correct.

rocco99, 2024-11-17:
I meant tft.fillScreen(TFT_BLACK) shows a white screen.

dave0003, 2024-11-17:
I'd recommend checking to make sure you're using the right User_Setup.h.  If your CYD version has 2 usb ports, you'll want to use the CYD2USB version.  I noticed something similar when I was using the wrong version.

https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/tree/main/DisplayConfig

rocco99, 2024-11-17:
Thanks. I'll try that out and let you know if it works

rocco99, 2024-11-18:
That did the trick. Thanks again.

rocco99, 2024-11-19:
It worked out well. Thank you. I made a simple calculator that I would like to share as a basic example. But I have never posted anything on github . How do I post my code so that others can benefit from it?

closed 2024-11-19 by rocco99