witnessmenow/ESP32-Cheap-Yellow-Display #278: difference R and non-R version of ESP32-2432S028

reinierdevl, 2024-12-08:
Note that there is a difference between the ESP32-2432S028R and the ESP32-2432S028 

My ESP32-2432S028, 2 USB version, has a ST7789 driver instead of the Ili9341

All the examples you provide are for the version with the Ili9341 version, so I asume the ESP32-2432S028R, and did not work form my CYD

I used bb_spi_lcd.h instead of TFT_eSPI.h as libray, used DISPLAY_CYD_2USB as device and that worked fine

paelzer, 2024-12-08:
In the "config" folder there is another folder called CYD2USB already containing a setup file for tft_espi and ST7789 which works fine with the 2 usb variants I had on my desk so far.

darknessblade, 2025-01-08:
I have a similar issue with the 2 USB version:

Model I bought: [Bundle deals]
https://aliexpress.com/item/1005007524336724.html?

Ili9341 / ESP32-2432S028R 
model code on the display: TPM408-2. B

I am using ESP-HOME, and tried various combinations for the pinout but none work, and I can't get it to display the "Hello world" test text, I even tried changing the driver/model settings to: "Ili9342"

slug, 2025-03-21:
@darknessblade 

2USB (micro + type-c) version:
```
spi:
  - id: spi_tft
    clk_pin: GPIO14
    mosi_pin: GPIO13
    interface: spi2
  - id: spi_touch
    clk_pin: GPIO25
    mosi_pin: GPIO32
    miso_pin: GPIO39
    interface: spi3

output:
  # Setup a pin to control the backlight
  - platform: ledc
    pin: GPIO21
    id: backlight_pwm

display:
  - platform: ili9xxx
    id: esp_display
    model: st7789v
    spi_id: spi_tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    invert_colors: false
    spi_mode: MODE3
    data_rate: 40MHz
    # data_rate: 2MHz
    # pixel_mode: 16bit
    color_order: bgr
    color_palette: 8BIT
    # update_interval: 1000ms
    # auto_clear_enabled: true
    dimensions:
      height: 240
      width: 320
      offset_height: 0
      offset_width: 0
    transform:
      swap_xy: true
      mirror_x: true
    # show_test_card: true

touchscreen:
  platform: xpt2046
  id: esp_touchscreen
  spi_id: spi_touch
  cs_pin: GPIO33
  interrupt_pin: GPIO36
  data_rate: 2MHz
  spi_mode: MODE0
  threshold: 400
  update_interval: 50ms
  transform:
    swap_xy: true
    mirror_x: false
    mirror_y: false
  calibration:
    x_min: 390
    x_max: 3800
    y_min: 210
    y_max: 3800
```

cropse, 2026-01-01:
@darknessblade have you tested it? I bought the same model you got, and I can't run the basic example, here is my code with @slug  suggestion, but it's still empty content :/
Example I refer: https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/tree/main/Examples/ESPHome/1-Helloworld

The code I try to run:
```yaml

# skip setup from captive_portal:

# ============================================================ 
# ESPHome Display related setup
# ============================================================
# Create a font to use, add and remove glyphs as needed. 
font:
  - file: ${font_direcotry}Arimo-Regular.ttf
    id: arimo20
    size: 20
    glyphs: "!\"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZÅÄÖ abcdefghijklmnopqrstuvwxyzåäö"

# Create a Home Assistant blue color
color:
  - id: ha_blue
    hex: 51c0f2

# ============================================================ 
# Home Assistant related setup
#
light:
  - platform: monochromatic
    output: backlight_pwm
    name: Display Backlight
    id: backlight
    restore_mode: ALWAYS_ON

# ============================================================ 
# Hardware related setup
#
# Setup SPI for the display. The ESP32-2432S028R uses separate SPI buses for display and touch
spi:
  - id: spi_tft
    clk_pin: GPIO14
    mosi_pin: GPIO13
    interface: spi2
  - id: spi_touch
    clk_pin: GPIO25
    mosi_pin: GPIO32
    miso_pin: GPIO39
    interface: spi3

output:
  # Setup a pin to control the backlight
  - platform: ledc
    pin: GPIO21
    id: backlight_pwm

# Setup the ili9xxx platform
#
display:
  - platform: ili9xxx
    id: esp_display
    model: st7789v
    spi_id: spi_tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    invert_colors: false
    spi_mode: MODE3
    data_rate: 40MHz
    # data_rate: 2MHz
    # pixel_mode: 16bit
    color_order: bgr
    color_palette: 8BIT
    # update_interval: 1000ms
    # auto_clear_enabled: true
    dimensions:
      height: 240
      width: 320
      offset_height: 0
      offset_width: 0
    transform:
      swap_xy: true
      mirror_x: true
    # show_test_card: true
    lambda: |-
      it.print(160, 140, id(arimo20), id(ha_blue), TextAlign::BOTTOM_CENTER, "Hello World!");
      it.rectangle(0, 0, 320, 240, id(ha_blue));
```