witnessmenow/ESP32-Cheap-Yellow-Display #284: esp32-2432S022C

Zucchy00, 2025-01-09:
New possible implementation or any way to make it work with esp32-2432S022C

NellowTCS, 2025-01-19:
I second this
here are images of it:

[image]

[image]
[image]
[image]

Zucchy00, 2025-01-30:
Yes exactly that, it would be great

CoolmdXi, 2025-02-15:
Have a look at bb_spi_lcd this board is supported in that library