witnessmenow/ESP32-Cheap-Yellow-Display #285: Using I2C and/or SPI

Jeronimo59, 2025-01-12:
Hello,

Has anyone tried to use I2C or SPI to communicate with external devices ? 
Which GPIO/connector is best to use ?

Thanks.

CoolmdXi, 2025-02-15:
i2c has been used by lots of people, not sure about SPi

phpo1-61, 2025-10-04:
Hye, I just managed to connect a NRF24L01 (SPI interface to a CYD ESP32-4827S043N.
I use LINUX OS, VScode and Platformio, and the HSPI port of ESP32.
My card is fitted with P2 connector labelled SPI (IO19,IO11, IO12 and IO13) and P3 connected. I use also IO18 on P3 connector.

Herafter the configuration:

 platformio.ini:
```c
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32-4827S043N]
platform = espressif32
board = esp32-4827S043N
framework = arduino
monitor_speed = 115200
monitor_rts = 0 
monitor_dtr = 0
monitor_filters = esp32_exception_decoder
lib_deps = nrf24/RF24@^1.5.0
```
Notice that the esp32-4827S043N.json file has been imported from site:
https://github.com/rzeldent/platformio-espressif32-sunton/
 and copied in local relative path:

.platformio/platforms/espressif32/boards

so that this board is known by platformio 

main.cpp: extracted from:

https://nrf24.github.io/RF24/index.html 

then Related Pages, Arduino, ESP32 example

```c
#include <SPI.h>
#include <RF24.h>
#include <printf.h>

// to use custom-defined pins, uncomment the following
int8_t MY_MISO = 11;
int8_t MY_MOSI = 13;
int8_t MY_SCLK = 12;
int8_t MY_SS =  19; // pass MY_SS as the csn_pin parameter to the RF24 constructor
int8_t MY_CE  = 18;
// notice these pin numbers are not the same used in the library examples
RF24 radio(MY_CE,MY_SS ); // the (ce_pin, csn_pin) connected to the radio

SPIClass* hspi = nullptr; // we'll instantiate this in the `setup()` function
// by default the HSPI bus pre-defines the following pins
// HSPI_MISO = 12
// HSPI_MOSI = 13
// HSPI_SCLK = 14
// HSPI_SS   = 15

void setup() {
  Serial.begin(115200);
  while (!Serial) {} //some boards need this
  printf_begin();
  hspi = new SPIClass(HSPI); // by default VSPI is used
  //hspi->begin();

  // to use the custom defined pins, uncomment the following
  hspi->begin(MY_SCLK, MY_MISO, MY_MOSI, MY_SS);
  if (!radio.begin(hspi)) {
    Serial.println(F("radio hardware not responding!!"));
    while (1) {} // hold program in infinite loop to prevent subsequent errors
  }
  // affichage type module nrf24
  if (radio.isPVariant()) {
    Serial.println(F("RF24L01+ detected"));
  } else {
    Serial.println(F("RF24L01 detected"));
  }
  radio.printPrettyDetails();
}

  // ... continue with program as normal (see library examples/ folder)

void loop() { 
  delay(5000);
}
```

Furthermore, I am interested in knowing which GPIO pins can be used to manage an I2C device on a CYD board: I can access through the connectors to these other pins (IO17, IO20)

Thank you in advance or your advice.