witnessmenow/ESP32-Cheap-Yellow-Display #290: Suggestion to Link My Related Project

mogrikid, 2025-01-26:
Hi Brian,

Thank you for your amazing project, on the CYD! I've found it incredibly helpful and inspiring.

I wanted to let you know that I’ve created a project based on yours, [PhilRadio](https://github.com/mogrikid/PhilRadio). In my project, I re-used an old Philips radio, used a CYD and made it ready for internet radio by reusing the old components of the radio. It features playing internet radio, persistent storage of the radio stations and a webserver to edit the stations.

If you think it could be useful to others, feel free to link it in your repository. I’d be honored to contribute in some way to the amazing work you’ve started! The code is far from perfect, but the radio runs flawless for hours and the audio quality is decent. Maybe others can build on it. 

Thanks again for all the effort you’ve put into this project!

Best regards,
Mogrikid

timmwille, 2025-02-08:
cool @mogrikid may I suggest you add an open source license for possible collaboration to your project? I would love to share it around as an example

mogrikid, 2025-02-09:
Hi @timmwille , 

thanks for the hint. I added the MIT license. I would be happy to see where you linked the project. 
Have fun!