witnessmenow/ESP32-Cheap-Yellow-Display #303: Esphome examples just give blank screen , error while compiling

nishithdev, 2025-02-24:
I have a CYD that is working fine for other projects, recently I was trying to install espHome and load example configurations from here 
https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/blob/main/Examples/ESPHome/1-Helloworld/yellowtft1.yaml

even for the basic ones like Hello World, I see the following error while flashing, https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/blob/main/Examples/ESPHome/1-Helloworld/yellowtft1.yaml

[16:27:42][C][ili9xxx:118]:   Invert colors: NO
[16:27:42][C][ili9xxx:121]:   => Failed to init Memory: YES!
[16:27:42][C][ili9xxx:123]:   Update Interval: 1.0s
[16:27:42][E][component:082]:   Component display is marked FAILED
[16:27:42][C][light:092]: Light 'Display Backlight'
[16:27:42][C][light:094]:   Default Transition Length: 1.0s
[16:27:42][C][light:095]:   Gamma Correct: 2.80
[16:27:42][C][captive_portal:089]: Captive Portal:

Has anyone faced similar issues?

Yes, I have added invert_colors: false.

edasque, 2025-03-03:
I get the same, any idea?

nishithdev, 2025-03-08:
thanks @edasque , this fixed the issue for me

closed 2025-03-08 by nishithdev