witnessmenow/ESP32-Cheap-Yellow-Display #309: 2USB variant only works with 1USB config

realflash, 2025-03-30:
I have what according to all the documentation in this repository is a CYD2USB. If I use the 1USB User_setup.h, HelloWorld example works as expected. If I use the 2USB User_setup.h, all I get is a white screen. It doesn't seem to be a case of inverted colours as I don't get black text on a white screen. Likewise if I invert the colours specified in the code I still get nothing.

Have I found a third variant, or have I misunderstood what inverted colours means?

realflash, 2025-04-24:
Well scratch that. With PlatformIO, I also get a white screen only with they cyd2usb target, and working hello world with the cyd target. Even though it has 2 USB ports. This device was bought through the links provided in this repository so I'm presuming it's the right one.

AlexD81, 2025-05-01:
> Have switched to platformio because backlight test wasn't working. With PlatformIO, I also get a white screen only with they cyd2usb target, and working hello world with the cyd target. Even though it has 2 USB ports. This device was bought through the links provided in this repository so I'm presuming it's the right one.
hi, im newbie with it, could you pls provide a sample  project in platformio.  Just need to get  working 'hello word' on screen orsomething like this, ,   i also have a version with 2 usb,   constantly getting white screen, and can't fix it

AlexD81, 2025-05-01:
> Have switched to platformio because backlight test wasn't working. With PlatformIO, I also get a white screen only with they cyd2usb target, and working hello world with the cyd target. Even though [...]

hi, im newbie with it, could you pls provide a sample  project in platformio.  Just need to get  working 'hello word' on screen orsomething like this, ,   i also have a version with 2 usb,   constantly getting white screen, and can't fix it

AlexD81, 2025-05-02:
i have 2USB variant, it works with User_Setup.h from 1usb version.   but touch is not working, looking for help