witnessmenow/ESP32-Cheap-Yellow-Display #318: Is there a zephyr configuration for CYD?

Karijn, 2025-05-27:


Karijn, 2025-05-30:
I figured out how to use lvgl on the CYD. It is probably not the best solution but it works.

(better solutions are welcome :-))

[cyb_lvgl.zip](https://github.com/user-attachments/files/20518219/cyb_lvgl.zip)

JoshuaRobertPreston, 2025-06-24:
new to CYDs and Zephyr, I've got hello world flashed ok and now trying to use your board configs - do I need to use patch or similar to generate the .yml files? Or I'll start hand crafting, any advice welcome.

JoshuaRobertPreston, 2025-06-27:
getting through my tutorials, your demo works great!
... had to understand about overlay files and specify:
west build -p always -b esp32_devkitc/esp32/procpu  -- -DDTC_OVERLAY_FILE=boards/esp32_cyd.overlay

LVGL and touchscreen input working, thankyou